jporter7/YCP-RT-ControlRoom #139: Improve Scale model

rvandemark, 2019-03-13:
1. I developed new Arduino code, I will continue to develop it and test it with the new PLC client code
2. I'll add "limit switches", and Kira suggested I just use little push buttons to simulate them, which I think is a good idea - so I'll be doing that
3. Improve ScaleModelPLCDriver to have as full of functionality as possible

rvandemark, 2019-03-17:
I’ve been working with David to make some of these changes, starting with how to position the push buttons to best represent the -2 to 362 limits. He’s been making himself more comfortable with the arduino code I updated back on 03-03-2019 too.

fdiaz3, 2019-03-18:
I've researched a few motors from the stock room, and I believe I found a set that will require very little if any code change. I will be putting them on by Wednesday of this week and making sure the model is more robust.