jporter7/YCP-RT-ControlRoom #175: Exception Handling / Unit Testing sprint, track activity here

rvandemark, 2019-03-23:
This is the one-and-only issue dedicated to tracking the activity of each of us during the EHUT sprint. We are aware of what classes we should each be extensively unit-testing (there's a picture in the control_room channel on Slack), so here, we should log which files we are modifying to add exception handling. Simply added a comment on here that contains a list of the files that you're adding exception handling to! Don't include files in the ControlRoomApplicationTests project.

**DUE DATE**: Your pull request should be in by 9PM on March 27th, PLEASE NO EXCEPTIONS (For Jason's sake, as he's the one handling pull requests).

rvandemark, 2019-03-23:
Modifying:
RadioTelescopeControllerManagementThread

tjones50, 2019-03-24:
Modifying:
CoordinateCalculationController

jporter7, 2019-03-25:
Modifying/testing:
- RTDbContext
- RFData
- ControlRoom
- RTDbConfiguration
- Configurations
- HeartbeatInterface
- SpectraCyberTestController

dhenry5, 2019-03-26:
PLCsafteyInterlockStatusConversionHelper
PLCradiotelescopestatusconversionhelper
PLCLimitSwitchStatusConversionHelper

closed 2019-03-31 by tjones50